PiSupply/Ryanteck #7: Traffic pHAT inconsistencies

lurch, 2020-10-01:
https://github.com/PiSupply/Ryanteck/blob/master/Traffic%20pHAT/img/Traffic-pHAT.jpg and https://github.com/PiSupply/Ryanteck/blob/master/Traffic%20pHAT/Hardware/schematic.pdf show that the board only has 3 LEDs.

https://github.com/PiSupply/Ryanteck/blob/master/Traffic%20pHAT/README.md and https://github.com/PiSupply/Ryanteck/blob/master/Traffic%20pHAT/Hardware/README.md say that it also has a button, a buzzer and an EEPROM ??

ryanteck, 2020-10-01:
Sorry, the Image & Schematic is correct. The description was copied from the Traffic HAT and I hadn't gone through yet to fully update.

ryanteck, 2020-10-01:
Would you like me to fix, or would you like to fix it if you're looking for PRs for hacktoberfest?

ryanteck, 2020-10-01:
Should now be fixed

closed 2020-10-01 by ryanteck

lurch, 2020-10-01:
I don't need any free tshirts, and even if I did I wouldn't need to go looking for "extra" PRs :wink: https://github.com/lurch

But anyway, thanks for fixing this Ryan :+1:

ryanteck, 2020-10-01:
No prob, thanks for opening the issue.