PiSupply/Ryanteck #11: C client source code

maggu2810, 2020-12-28:
Hi,
thank you for this board, the specifications and the source code.

You provide a python module to access the board (let's say a client python library).
The firmware for the board is written in C, so there is also the (let's name it server) C code.

Do you consider to provide the client library source code also for C / C++?
IMHO the code itself could be the same for all platform, we need just a simple wrapper for the serial port handling. This wrapper needs to be implemented for the specific OS.
It would be good to provide the code base here at a central place. Otherwise it needs to be developed by everyone itself.

ryanteck, 2020-12-28:
There is no C code for the computer side of the device.

maggu2810, 2020-12-28:
I know that there is currently no one. So the request if you could add one.
But if there is no interest to provide it on the upstream project, no problem. I just thought it would be a good idea.
I will write it myself.

ryanteck, 2020-12-28:
Sorry, I read your request as an assumption that I had written a C client library just because the microcontrollers was written in C.

However currently there's no plans to write a C library as python is targeted due to it's popularity as a language used on the Pi.

shawaj, 2020-12-28:
@maggu2810 all of the software and firmware is up on GitHub already as open source.

we welcome any PRs on the repo so if you are willing to write the code we can definitely look at including it. But we will not write one ourselves at this point because there is not any demand for it - you are the first person to ask about it.

maggu2810, 2021-01-31:
The first version could be found here: https://github.com/maggu2810/librtkgpio
I will update the library if there is any need.

shawaj, 2021-01-31:
@maggu2810 great we will take a look. Cc @ryanteck 

I'll add a link also to the repo