PiSupply/Ryanteck #14: This library crashes when imported

cnobile2012, 2022-02-26:
I'm using Python 3.8.10. I get the following traceback.

```
In [1]: from RTk import GPIO

Error: RTk.GPIO not detected.
For more support please visit our website at http://Ryanteck.com/rtk-000-00C
Press enter to close.

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Input In [1], in <cell line: 1>()
----> 1 from RTk import GPIO

File ~/.virtualenvs/RTk_Test/lib/python3.8/site-packages/RTk/GPIO.py:54, in <module>
     43 from RTk import adaptors
     46 #from os import sys, path
     47 #thisdir = path.dirname(path.abspath(__file__))
     48 #sys.path.append(thisdir)
   (...)
     51 
     52 #Temporarily changing back to normal serial
---> 54 from RTk import rtkserial
     56 instance = protocol.GPIOClient(adaptors.SerialAdaptor(rtkserial.s), DEBUG)
     58 def setwarnings(option):

File ~/.virtualenvs/RTk_Test/lib/python3.8/site-packages/RTk/rtkserial.py:36, in <module>
     34         print ("\nError: RTk.GPIO not detected.\nFor more support please visit our website at http://Ryanteck.com/rtk-000-00C\nPress enter to close.")
     35         input()
---> 36         exit()
     38 #print ("Debug cable fully detected fine.\n Press enter to launch the terminal.")
     41 BAUD = 230400

NameError: name 'exit' is not defined
```

There are two issues in that traceback.
  1. Error: RTk.GPIO not detected.
  2. The exit() function is not found. This is correct there is NO built-in function named exit().

I also found that there are tabs used in the code which go against PEP8 the Python programming style guide.

I've been developing python code for over 20 years, so I know these are real issues.

tvoverbeek, 2022-03-06:
How did you install the library? Using the install via setup.py does not work (Probably a victim when the Ryanteck code was moved to PiSupply).
I just tried with pip on my Mac (python 3.9.1 from homebrew) and it works fine.
No import error.

Regarding exit(). Yes, it is not recommended, better to take your proposed approach.
I am not the maintainer for the Ryanteck code, nor working for PiSupply, hence I cannot guarantee that it will be fixed.

cnobile2012, 2022-03-06:
Thanks for your response. I also used pip to install it in a Python virtual environment. As it turns out this repo is no longer active. To find the active repo go to https://bitbucket.org/ryanteckltd/rtk-gpio-python-side/src/master/.

What I did was to copy the code from the pip install and fix the code myself, expecting that the code will never be maintained even in the newer Bitbucket repo. I may eventually push my fixes to my GitHub account if I decide to do anything other than a few bug fixes. There is little support for a lot of features the board can do, but the software cannot do.